staticintlucas/goldfish #3: "Stole" your schematic -- designed a new board

JeremyJStarcher, 2020-11-30:
Just letting you know that I snagged a copy of your schematic and used it as the base of a pin-compatible board, but with different design goals.

Right now my project is not an official fork of yours, but that should be fairly easy to do.

If you'd rather I changed the name of my project because it has different goals than yours, I'm more then willing to do that as well.

Of course, either way, full credit will be given.

https://github.com/JeremyJStarcher/Goldfish-clone-JLCPCB

staticintlucas, 2020-11-30:
The design is licensed under CC BY-SA, so as long as you give credit and license under the same terms it's all good. You can add the copyright info to your README like I have, but you already have it in the schematic so I'm not sure if that's even necessary.

I don't mind you using Goldfish in the name at all, as long as you don't pretend that yours is the 'original' Goldfish. 

If you would like to make your project an 'official' fork feel free, I really don't mind either way. We could also add it to this repo as a new branch making it an official Goldfish variant, but if you'd rather keep it in your own repo that's fine too. The choice is up to you.

JeremyJStarcher, 2020-12-01:
Thanks for the response, really appreciate it.
While I know that the file was under the CC BY-SA license, I consider it good form to contact a project owner when work is modified, so thank you for the response.

I meant to have the licence there in the README, twas just an oversight that I have corrected.

I'll wait until I get the boards back and make sure they actually work and then I will consider if I want to make things a fork of your repo or if I want to put in a pull request.

Alas, the maker is out of 32u4 at the moment, so I have to wait for a resupply before I can put in my order and make sure that everything is oriented properly.