staticintlucas/goldfish #4: Board from JLCPCB not working

vicgwee, 2020-12-02:
First I would like to say thanks for making such a neat board, I really like it! 

For a keyboard project I went to make 20 of these on JLCPCB with their PCB assembly service. I couldn't find the original crystal oscillator, so I had to change it to [this](https://datasheet.lcsc.com/szlcsc/Yangxing-Tech-X322516MLB4SI_C13738.pdf). 

When the boards came, I soldered the original Hirose connector on but I couldn't get my PC or Mac to detect it at all via USB. I thought perhaps I needed to burn a bootloader, but I'm having trouble using a Nano as an ISP to flash it. 

Here is the board, the solder work by JLCPCB and the traces look great. 

[image]

laughedelic, 2020-12-02:
relevant: https://github.com/staticintlucas/Goldfish/issues/3

jankolkmeier, 2020-12-02:
I'm you want the *-AU* version of the atmega, the *-MU* version as you suspected comes without bootloader ([see here](https://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-7766-8-bit-AVR-ATmega16U4-32U4_Summary.pdf)), so you'll definitely need to flash the bootloader.

How exactly did you try flashing it the bootloader so far?

jankolkmeier, 2020-12-02:
please double check for your self, but this is how I think you should connect the ISP headers:
[image]

staticintlucas, 2020-12-02:
It looks like something went wrong when exporting your gerbers for JLCPCB. I'm assuming you generated your own gerbers and didn't use the ones I released [here](https://github.com/staticintlucas/Goldfish/releases/tag/revC)?

If you compare the released gerbers (screenshot from gerblook.org) to your board you can see the vias that are meant to connect the grounds front and back aren't actually connected to the ground planes on your board.

[image]

The only real solution I can think of is to peel back some of the soldermask in these areas and bridge the gap between the vias and the plane with some solder. That's a tedious task though for 20 boards. 😧

vicgwee, 2020-12-03:
Oh no that looks bad! Yes I had to generate my own gerber because I was panelising it together with my keyboard PCB.

Thanks so much for the help, I'll see if I can get it to work.

closed 2020-12-03 by vicgwee

fi0, 2020-12-04:
@vicgwee did you find out what was causing the gerber errors?