staticintlucas/goldfish #5: Question a

marekpiechut, 2021-05-17:


marekpiechut, 2021-05-17:
Hi, sorry for opening an issue with a question, but could not find a better place to ask.

I'm an electronics newbie, so it might be a dumb question, but I cannot figure it out. Why are you using 4 x 0.1uF capacitors on VCC and 0.1uF and 4.7uF capacitors on VBUS?

When going through the datasheet for Atmega32u4 I've found that they recommend a 10uF on VBUS (so maybe you have smaller capacitance to improve reset time on reconneciton?). There's nothing about capacitors on VCC.

I've also checked ProMicro schematics, and they have a 1uF on VCC and VBUS connected together.

Sorry if it's a dumb question, but I'm looking into it for few hours already and cannot figure it out.

staticintlucas, 2021-05-17:
Hey,

The 0.1uF caps are what's known as decoupling capacitors. You can Google the term for more info, but the TL;DR is they're small caps placed close each power pin on a chip. They allow any high frequency transients or noise in the power plane to pass straight to ground so that the voltage received by the chip itself remains stable. It's a common practice for any IC, not just microcontrollers. I'm actually surprised the ATmega datasheet doesn't mention this, although I'm sure it is mentioned in some application note somewhere.

So in the schematic there is an 0.1uF cap for each of the VCC and AVCC pins, and one on VBUS for the UVCC pin. The 4.7u is for the VBUS pin. I'm not sure exactly why I didn't use 10u, it's possible I just happened to have some extra 4u7 caps around when I first designed this. For all these caps the exact value doesn't really matter too much, as long as it's the same order of magnitude it'll work just as well.

On the Pro Micro they have a 1uF on VBUS rather than 10u; I have no idea why, but I can't imagine it makes a huge difference. For VCC they use an LDO to step down the incoming 5V USB power to 3.3V, and you can see quite a few capacitors around the LDO to minimise any transients. I guess that's why they decided they didn't need decoupling caps, although it's usually good practice to keep them anyway.

I hope this has answered all you questions! And feel free to ask if you have any more 😄

marekpiechut, 2021-05-18:
I've did some reading about the subject and now it makes sense. You have 4 VCC pins on the chip, so you have 4 capacitors near it.
Learned something new :) Thanks 🙏.

closed 2021-05-18 by marekpiechut