staticintlucas/goldfish #7: Me brain not so good, gerber file how?

geenoee, 2021-12-25:
I'm relatively new to CAD and all this stuff, I decided I wanted to commission a PCB so I can handwire my keyboards. I've downloaded Gerber files before but I'm having trouble specifically with this project in which my Gerber files when I plot them on KiCAD. don't actually download as Gerber files and as some other type of file. With that said the PCB manufacturer can't support those types of files. If you can, can you either put a zip of the Gerber files or explain to me what happened? pretty please and thank you for creating this masterpiece of a microcontroller. I don't know if this is the right place to talk about this but sorry if it isn't

staticintlucas, 2021-12-25:
The source files in the repo are KiCad design files. If you want Gerbers you can click on Releases on the right hand side and download the .zip with Gerber files.

geenoee, 2021-12-25:
> The source files in the repo are KiCad design files. If you want Gerbers you can click on Releases on the right hand side and download the .zip with Gerber files.

thank you so much!!!!!!

closed 2021-12-25 by staticintlucas