staticintlucas/goldfish #8: Designing a 31-m-13 usb-c version

JiaGengChang, 2022-05-13:
I was building my own controller using the latest release, but realised that I do not have the Hirose type CX70M usb-C connector.  In my region, shipping of this kind of connector is quite expensive.

Hence, I am working on modifying the usb-C footprint to HRO 31-M-13, which can be purchased from aliexpress (see https://www.aliexpress.com/item/32998900371.html?spm=a2g0o.order_list.0.0.31b61802zlAvym for example). Will let you know how it goes.

staticintlucas, 2022-05-14:
Nice idea. I had been planning to redo the PCB with another connector at some point. The availability of different connectors has changed a lot since I first designed this.

If you open source your version let me know. I'll link it in the readme so others can find it more easily.

JiaGengChang, 2022-05-15:
I have made the made the modification (along with a two other minor additions).
The link is https://github.com/JiaGengChang/Whale.
I actually ended up using another USB-C connector which is the one I actually provided in the link above. Its part number on LCSC.com is C168688.
Would appreciate if you could give my adapted version a mention.

staticintlucas, 2022-05-15:
Added a link to it in the README in 2be9e98

JiaGengChang, 2022-05-16:
Thanks a lot

closed 2022-05-16 by JiaGengChang